mkuthan/solis-cloud-control #13: slot1_charge_time provides state 00:01-2300:5000 which is too long (maximum length 11)

wjaszczuk, 2025-04-09:
Hello!

Thank you for your work!
I've tried to use solis-sloud-control. I've used the control API from Solis-Sensor, but I gave in. My datalogger stopped responding, and I had to reset it plus disable control integration from the solis-sensor.

Now, I'm trying to automate selling energy when the price is the best, and I've encountered some errors. I think it's something from Solis's side.
When I tried to change the charging slot I got this kind of error:

```
slot1_charge_time provides state 00:01-2300:5000 which is too long (maximum length 11)
```

I think the 2300 value comes from here:
```
{
        "msg": "00:01-2300:5000",
        "yuanzhi": "00:01-2300:5000",
        "sysCommand": {
          "id": "5946",
          "createDate": 1704853050000,
          "updateDate": 1736216423000,
          "remarks": "\u5145\u653e\u7535V2",
          "deleteFlag": 0,
          "inverterModelId": "2",
          "widgetType": 201,
          "functionCode": 16,
          "registerAddr": 43711,
          "functionCodeRead": 3,
          "registerAddrRead": 43711,
          "registerNumber": 4,
          "value": "1,1,1,1",
          "productModel": "c7,c8,c9,c7,c8,c9,ca,cb,cc,3354,3355,3356,3357,3358,3359,f2,f3,f4,f5,f6,f7,f8,3101,3102,3103,3104,3105,3106,3107,3108,3109,3110,cb,cc,3354,3355,3356,3357,3358,3359,f2,f3,f4,f5,f6,f7,f8,3101,3102,3103,3104,3105,3106,3107,3108,3109,3110,3111,3112,3113,3114,3115,3121,3122,3123,3124,3125,3155,3127,3128,3129,3130,3160,3161,3162,3116,3117,3118,3119,3120,3126,3127,3128,3129,3130,3160,3161,3162,3130,3160,3161,3162,3131,3132,3133,3134,3135,3136,3137,3138,3139,3140,3141,3142,3143,3144,3145,3146,3147,3148,3149,3170,3171,3172,3173,3174,3201,3202,3203,3204,3205,3206,3207,3208,3209,3211,3212,3213,3214,3215,3313,3314,3315,3316,3317,3318,3322,3301,3302,3303,3304,3305,3306,3307,3313,3314,3315,3316,3317,3318,3322,3330,3331,3332,3333,4305,4306,4307,4308,5120,5121,5124,5125,5122,5123,5300,5301,5302,5303,5304,5305",
          "pid": "6209",
          "name1": "\u5145\u7535\u65f6\u6bb51",
          "name2": "Charge Time Slot 1",
          "order": 101,
          "isJiChu": 0,
          "needRead": 0,
          "bit": -1,
          "unbit": "-1",
          "othersOnbit": "-1",
          "xiaoshu": -1,
          "auth": 0,
          "edit": 0,
          "batchCommand": 0,
          "offSet": -1,
          "isPageShow": 0,
          "isApiShow": 0,
          "dspMin": 0,
          "dspMax": 65535,
          "hmiMin": 0,
          "hmiMax": 65535,
          "standardNotShow": "",
          "buttonShow": 0,
          "buttonControlShow": 0,
          "on": 1,
          "off": 0,
          "passWordCheck": false,
          "passWordCheckType": 0,
          "needRefresh": 0,
          "freshByRules": 0,
          "idStr": "5946"
        },
        "command": "01 03 a8 02 00 0f 84 6e;01 03 a8 42 00 25 04 65;01 03 a9 d9 00 01 74 6d;01 03 aa bb 00 32 95 e2",
        "cid": "5946"
      },
```

```
2025-04-09 08:31:52.846 DEBUG (MainThread) [custom_components.solis_cloud_control.coordinator] Data read from API: {160: '7', 7963: '100', 158: '10', 7229: '10', 157: '10', 5948: '0', 5928: '0', 5916: '0', 5946: '00:01-2300:5000', 5917: '0', 5918: '0', 5919: '0', 5920: '0', 5921: '0', 5967: '0', 5965: '0', 5922: '1', 5964: '08:00-09:00', 5923: '0', 5924: '0', 5925: '0', 5926: '0', 5927: '0', 499: '20', 636: '51'}
```

All my inverter time charging and discharging settings are set to 00:00 - 00:00 except discharging slot 1, which is set to 08:00-10:00. I see these values both on the inverter and in the Solis official app in the control menu.

Could you help me?

Thanks, Wiktor

mkuthan, 2025-04-10:
Thanks for detailed logs, it helps a lot. 

Value '00:01-2300:5000' comes from Charge Slot 1 (CID = 5946), try to update this slot using Solis app, for example from 00:00-00:00 to 00:00-01:00 and then back to 00:00-00:00. I know, it looks like a workaround but it could help :)

In my installation I've got such error only once, but it looks that such situations are not exceptional.

Don't close the issue even if my workaround proves to be effective, please. I'm going to figure out how to handle such erronous API responses in the integration.

wjaszczuk, 2025-04-10:
Hi!

Thank you for your advise. I've set all the charging and discharging slots and after that I checked the HA integration. Unfortunately I still get the same error.
Thanks!

mkuthan, 2025-04-11:
Update the integration to version 1.3.1 please :)

wjaszczuk, 2025-05-01:
Hello @mkuthan. I've read another topic from the repository issues, and I asked Solis to upgrade my firmware. Now, everything works like a charm. I recommend using their support desk and asking for an upgrade. I waited around 3 days, and they did it remotely.

I appreciate your support!

PR #14 (2025-04-11): fix: validate time slot returned by API
If API returns invalid data for charge or discharge time slot, ignore and log warning with details.

Closes #13
closed 2025-05-10 by mkuthan