mkuthan/solis-cloud-control #16: code B0115 returned during initialisation

astronaut63, 2025-04-12:
Homeassistant has been running with the Solis Sensor addin for many tears
I tried adding this integration (as I could not get the recently added controls to work in the original)
Initialisation fails:
2025-04-12 17:27:12.799 DEBUG (MainThread) [custom_components.solis_cloud_control.api] API request '/v2/api/atReadBatch': {
  "inverterSn": "160Fxxxxxxxxxxxx",
  "cids": "160,7963,158,7229,157,5948,5928,5916,5946,5917,5918,5919,5920,5921,5967,5965,5922,5964,5923,5924,5925,5926,5927,499,636"
}
2025-04-12 17:27:12.962 DEBUG (MainThread) [custom_components.solis_cloud_control.api] API response: {
  "success": true,
  "code": "B0115",
  "msg": "Sending failure ,the current datalogger is offline or disconnected",
  "data": null
}
2025-04-12 17:27:12.962 WARNING (MainThread) [custom_components.solis_cloud_control.api] Retrying due to error: API operation failed: Sending failure ,the current datalogger is offline or disconnected (API response code: B0115) (attempt 1/5)

API manual gives code B0115 as "fail in send"

Have also tried disabling and removing the original Solis Sensor add-in - still get the same error

Remote control via app and website for charging times still works

My inverter is nearly 4 years old - still has just 3 charge/discharge slots
Version Info: Looking at the real Inverter Screen this is shown as 3C002A - so HMI 3C and DSP 2A
The same is shown on the Soliscloud Inverter screen ( Inverter Version 3C002A)

mkuthan, 2025-04-13:
According to the API documentation, code B0115 is described as "fail in send," though its exact meaning is unclear.

The issue with the slots is likely due to the fact that, even if the integration exposes only one charge and one discharge slot, it still reads the status of all 2x6 slots. To resolve this, it's necessary to construct the request to specifically switch on/off the first slots.

Could you contact Solis support (https://solis-service.solisinverters.com/) to request a firmware update for your inverter?

In the meantime, I plan to modify the atReadBatch operation to send requests only for enabled entities. It could help when you disable all entities related to the time slots.

mkuthan, 2025-04-13:
You could try fix/workaround from #18. Integration should start but the slots will not be available.

mkuthan, 2025-05-28:
@astronaut63 @diricxbart - could you try [v2.4.0-beta.1](https://github.com/mkuthan/solis-cloud-control/releases/tag/v2.4.0-beta.1)?

mkuthan, 2025-05-29:
I have one more idea, orderid in the response can be used for the next read request. This is a documented feature, but i didn't explore this path. Calls to my inverter don't return orderid. With debug logs from your inverter i could try to implement such re-read logic in the integration.

diricxbart, 2025-05-29:
I'd suggest awaiting the behavior after the firmware update, hopefully exposing the atReadBatch as well

astronaut63, 2025-05-29:
Hi, I had my firmware upgraded about 6 weeks ago by Solis (now version = "4A00B0", previously  3C002A )
I've tried the new beta and I still see some debugmessages -  the B0115 error code for CIDs 15

2025-05-29 12:17:10.735 DEBUG (MainThread) [custom_components.solis_cloud_control.api.solis_api] API request '/v2/api/atReadBatch': {
  "inverterSn": "160F3xxxxxxxxxx",
  "cids": "15"
}
2025-05-29 12:17:10.843 DEBUG (MainThread) [custom_components.solis_cloud_control.api.solis_api] API response: {
  "success": true,
  "code": "B0115",
  "msg": "Sending failure ,the current datalogger is offline or disconnected",
  "data": null

However the inverterdetail call is returning a large set of parameters (I don't think this was working before)
I'm also seeing "[custom_components.solis_cloud_control.inverters.inverter_factory] Unknown inverter model 'F6',"
F6 is the model in the inverter details but I am 50/50 as to whether that changed when the firmware was upgraded as the model number changed at that time (now RHI-6K-48ES-5G  (was RHI-5K....)

I'm also going to raise a new ticket as I haven't been able to manually change the charge slots using either app or website since the firmware change (and it was only working 1% of the time since March 2025 )

diricxbart, 2025-06-05:
I reached out to have the firmware of my inverter updated. Solis indicated this was not possible with my DataLogger version (DLS-W WiFi) and suggested to replace it with the more recent S2-WL-ST v1.4 Solis WiFi/LAN Data Logger.

After having done this update, the initialization issues I was experiencing with this Solis Cloud Control integration are automagically resolved, the entities show up in Home Assistant !

Solis will probably still update the inverter firmware to the latest version, fingers crossed all goes well...

Thanks a lot for your help!

MijnSpam, 2025-06-14:
I'm getting the same error, my inverter doesn't have any battery function.
Type: Solis-3P10K-4G
But i'm hoping the to use the plugin later to limit the generation power in the future.
In the APP under inverter control i can read and set settings.