mkuthan/solis-cloud-control #22: S6-GR1P2.5K - Max. Export power

PeterKawa, 2025-04-24:
Hi!

Is it possible to add suppprt for the S6-GR1P2.5K inverter? I'd like to be able to control the maximum export power only.

I know it's not supported, but it got recognised and I could add it to the integration. Only the entities can't be populated.
I tried a few times, when the sun was shining, but:
(`Failed setup, will retry: API operation failed: Communication error. Please refresh and try again later (API response code: 1)`)

I can control the export power when I use the API tester https://soliscloudapitester.netlify.app/

[image]

When you're feeling like adding it, 
Anything I can do to help? 

Best, Peet

mkuthan, 2025-04-25:
It seems that regular PV inverters (like yours) and hybrid inverters (like mine) use distinct sets of commands. Refer to the following document for details:
[Command List - SolisCloud Control API](https://oss.soliscloud.com/doc/SolisCloud_control_api_command_list.xlsx)

I am unsure how to determine which set of commands a specific inverter supports.
Without access to the PV inverter, implementing and testing such a change would be challenging.

In general, I envision the following logic for the coordinator:

inverter_detail = api.inverter_detail() # new method

if inverter_detail.is_pv_inverter: # how to implement this logic?
    # Use the PV-specific set of CIDs (your case)
elif inverter_detail.is_hybrid_inverter:
    # Use the existing set of CIDs
else:
    raise Error("Unsupported inverter type") # or better filter out unsupported inverters in the config flow

Additionally, you will need to determine how to dynamically adjust the list of supported entities. I suggest not reusing the same entity for different types of CIDs, even if they serve similar purposes for PV and hybrid inverters. Currently, the integration disables entities if a CID for the given inverter type does not exist (e.g., charge/discharge slots). I'm not sure if such design is good enough for this change.

PeterKawa, 2025-05-15:
Thanks for looking into it, Marcin!
I've installed v2.0.0-alpha.3, it recognises my inverter, but I'll have to wait for sunshine tomorrow for the inverter to be online.

What happened in the meantime:
This very recent PR [#463](https://github.com/hultenvp/solis-sensor/pull/463) of the SolisCloud Sensor integration v3.11.0 works well: it's now possible to turn the inverter on/off, and 0-110% power setting works using the API.

mkuthan, 2025-05-16:
Thanks for the screens!

Let’s give [v2.1.0-alpha.1](https://github.com/mkuthan/solis-cloud-control/releases/tag/v2.1.0-alpha.1) a try.

I've added explicit support for your inverter (and introduced string inverter "abstraction") because, for the hybrid inverter, cid: 15 returns a nonsensical value like "493.93". It doesn’t match anything real. If it works this way, it will open up new and interesting possibilities.

PeterKawa, 2025-05-16:
Sorry for the late reply. Too late to test your update with daylight.
v2.1.0-alpha.1 now errors with B0072, "device is offline", which is absolutely correct atm 😉 To me that sounds a lot better than "communication error" hehe.
Be back tomorrow!

[image]

mkuthan, 2025-05-17:
Any other CID to add for your inverter?

PeterKawa, 2025-05-17:
Powering On/Off would be nice!
CID:48 (on) - value to send: 190
CID:53 (off) - value to send: 222

I could turn it on/off by using those particular values I've found by requesting the readSettings:

[image]
[image]

mkuthan, 2025-05-17:
here you have a new version with on/off switch - [v2.1.0-alpha.2](https://github.com/mkuthan/solis-cloud-control/releases/tag/v2.1.0-alpha.2)

PeterKawa, 2025-05-18:
Working like a charm, Marcin!
Much appreciated!

closed 2025-05-18 by mkuthan