mkuthan/solis-cloud-control #24: Maximum export power for RHI-3P10K-HVES-5G

arjanboer, 2025-04-29:
Hi! Fantastic project!

I integrated this project with my Home Assistant to manage the maximum export power dynamically. My goal is to adjust the export limits based on market prices—keeping it as low as possible when export prices are negative, and raising it when prices are favorable.

Using the Max Export Power entity, I can successfully update the value on my inverter. However, I’ve noticed that the values are scaled by a factor of 100. For example, while my inverter’s default maximum export power is 10,000 watts, the entity in Home Assistant displays it as 1,000,000.

When I attempt to update the entity with a value that is 100 times higher (e.g., 1,000,000), I encounter an error stating the maximum allowed value is 13,200. Setting the value to 13,200 translates to just 132 watts on my inverter.

Could this be a bug in the code, or is it possible that my inverter operates differently than expected?

mkuthan, 2025-04-30:
Scaling by a factor of 100 is implemented by design. It's required at least for S6-EH3P12 hybrid inverter. I'm going to fill Solis support ticket with question if this is by design or I have buggy firmware.

mkuthan, 2025-05-10:
@arjanboer I've just published the alpha version of the integration prepared to handle different inverter types. May I kindly ask you to test it? :blush:

1. **Get the alpha release**: https://github.com/mkuthan/solis-cloud-control/releases/tag/v2.0.0-alpha.1
2. **Important Note**: After the initial run, all existing entities could disappear, but don’t worry—it should be easy to fix! :sweat_smile:
3. **Check Your Inverter Model**: The integration displays the model as shown below (3331):
  > [image]
4. **Configuration Update**: Open the [inverter_factory.py](https://github.com/mkuthan/solis-cloud-control/blob/v2/custom_components/solis_cloud_control/inverter_factory.py) and add the configuration for your inverter. Start with an exact copy of `_create_3331` function and modify the `InverterMaxExportPower` entry by removing the scale attribute.
5. **Feedback Request**: What do you think about this framework for adding support for new inverters? :thinking_face:

Thanks! :blush:

arjanboer, 2025-05-10:
It works! My inverter is 'CA' so I added:     `elif inverter_info.model == "CA":
        return await _create_ca(api_client, inverter_info) `

Set the scale to 1 and power limit to 10000. This framework should work fine.

jesperkroghc, 2025-05-23:
I have also a max export power error related to my Inverter when trying to set a value. A grey box pops up showing fx - . Failed to set value for CID 499. Expected: 103, got: 108

For details see #32 - now on 2.2.2

mkuthan, 2025-05-24:
> I have also a max export power error related to my Inverter when trying to set a value. A grey box pops up showing fx - . Failed to set value for CID 499. Expected: 103, got: 108
> 
> For details see [#32](https://github.com/mkuthan/solis-cloud-control/issues/32) - now on 2.2.2

1. Create separate issue please
2. Provide debug logs
3. Provide diagnostic data (download diagnostic button)
4. Show your inverter state regarding max power before and after change. What do you see in the Solis App?

closed 2025-07-01 by github-actions[bot]