mkuthan/solis-cloud-control #29: Support Request: Inverter S5-GR3P12K

josadmiraal, 2025-05-18:
Thank you for your interest in expanding the capabilities of this integration! To help us add support for your new Solis inverter, please provide as much of the following information as possible.

Sample device info page (use "Preview" tab to see the image):

[image]

**1. Inverter Model Name/Type:**
Please provide the exact model name/type of your Solis inverter (e.g., S6-EH3P12K02-NV-YD-L).
S5-GR3P12K
```
[Your Inverter Model Name/Type Here]
```

**2. Inverter Model Identifier/Number:**
Please provide the model identifier/number of your Solis inverter (e.g., "3331").
0507
```
[Your Inverter Model Identifier/Number]
```

**3. Diagnostic Information:**
Download diagnostic from Home Assistant integration page. Please paste `inverter_info` and `coordinator_data` fields within the code block below.

```json
[Paste Your Diagnostic JSON Here]
```

**4. Additional Details:**
Please provide any other relevant information that might help us add support for this inverter. For example:

- What specific features or sensors are you hoping to see?
- Logs or error messages that you have encountered
- Any other context or observations

We appreciate you taking the time to provide this information. Adding support for new hardware can be complex, and your detailed input is crucial for success!

josadmiraal, 2025-05-18:
I have this inverter. but i get a Unknown inverter model '0507' in the home assistant log.
can you tel me how to get my inverter compatible.

mkuthan, 2025-05-18:
@josadmiraal - I added support for your inverter but don't expect more than limiting export power and inverter on/off switch.

Could you make me a pleasure and share diagnostic info from your inverter (just click button "Download Diagnostic" on the device page). I need entries from "inverter_info" field only.

Try this version: [v2.2.0-alpha.1](https://github.com/mkuthan/solis-cloud-control/releases/tag/v2.2.0-alpha.1)

Cheers!

josadmiraal, 2025-05-19:
Thanks Marcin. 😊

I just want to switch my inverter on/off via home assistant.
I installed version 2.2.0-alpha.1 and it works perfect.
Here is my diagnostic download.

PR #30 (2025-05-20): feat: add support for S5-GR3P(3-20)K
Closes #29
closed 2025-05-20 by mkuthan