mkuthan/solis-cloud-control #31: Support Request: [Inverter Model Number/Type]

japers, 2025-05-20:
Thank you for bringing in the control functionality!

It would be great to be able to read the switch status but I note from other posts this is complicated. Is there perhaps a way to simply read whether "Grid Charging" is allowed?

(I have two inverters in parallel and the Solis Modbus integration is unreliable for me. Sensors will report unavailable from time to time for a few minutes to hours. It works fine if I only connect one inverter.  I was hoping this would allow a more reliable way of reading/controlling charging.)

mkuthan, 2025-05-20:
What is a "switch status"? Could you provide CID for this operation by scrapping Solis web app traffic using dev tools?

"Allow Grid Charging" is already available as Storage Mode attribute: https://github.com/mkuthan/solis-cloud-control/blob/e960fd1f69b58bba8ec1aca80ecc0174a0a5cb73/custom_components/solis_cloud_control/select.py#L83

mkuthan, 2025-05-24:
You can read the status from "allow_grid_charging" attribute of "storage_mode" entity, see HA [templating documentation](https://www.home-assistant.io/docs/configuration/templating/#attributes).

[image]

Although there is no way to change the entity attribute in HA, this is a read-only property AFAIK. 

I see two potential solutions:

1. Create regular entities/switches as you proposed, not easy but feasible. The switches could be disabled by default as features for more advanced users.
2. Create dedicated services, [doc](https://developers.home-assistant.io/docs/dev_101_services). They would be available mainly for scripts and automations. Sample call will look like:

```
service: solis_cloud_control.set_storage_mode
data:
  mode: "Self-Use" (mandatory)
  allow_grid_charging: true (optional)
  battery_reserve: false (optional)
  ... any other attributes encoded as final storage mode
```

Let me know what do you think :)

closed 2025-05-28 by mkuthan

japers, 2025-05-29:
Thank you for this!! Am currently away but will use this as soon as I am back :-)

PR #34 (2025-05-28): feat: add BatteryReserve and AllowGridCharging switches
Closes #31