mkuthan/solis-cloud-control #32: Support Request: [3306/S6-EH3P10K-H]

jesperkroghc, 2025-05-20:
Thank you for your interest in expanding the capabilities of this integration! To help us add support for your new Solis inverter, please provide as much of the following information as possible.

**1. Inverter Model Name/Type:**
Please provide the exact model name/type of your Solis inverter (e.g., S6-EH3P12K02-NV-YD-L).
Machine Type S6-EH3P10K-H

**2. Inverter Model Identifier/Number:**
Please provide the model identifier/number of your Solis inverter (e.g., "3331").
Model 3306

[Your Inverter Model Identifier/Number]
Inverter Version 150009-050004

**3. Diagnostics Information:**
Download diagnostics from Home Assistant integration page. Please paste `inverter_info` and `coordinator_data` fields within the code block below.

Inverter 1033060233210147
[1 enhed](http://homeassistant.local:8123/config/devices/device/12db043148ae49c3b4e1c5f964e763fc)
Opsætning mislykkedes - prøver igen: API operation failed: The necessary parameters are empty or invalid (API response code: I0000)

```json
[Paste Your Diagnostic JSON Here]
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2025.5.2",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.13.3",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Copenhagen",
    "os_name": "Linux",
    "os_version": "6.6.74-haos-raspi",
    "supervisor": "2025.05.1",
    "host_os": "Home Assistant OS 15.2",
    "docker_version": "28.0.4",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "solis": {
      "documentation": "https://github.com/hultenvp/solis-sensor/",
      "version": "3.11.0",
      "requirements": [
        "aiofiles>=23.1.0,<25.0.0"
      ]
    },
    "frigate": {
      "documentation": "https://github.com/blakeblackshear/frigate",
      "version": "5.9.2",
      "requirements": [
        "hass-web-proxy-lib==0.0.7"
      ]
    },
    "hacs": {
      "documentation": "https://hacs.xyz/docs/use/",
      "version": "2.0.5",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "solis_cloud_control": {
      "documentation": "https://github.com/mkuthan/solis-cloud-control",
      "version": "2.2.0",
      "requirements": []
    }
  },
  "integration_manifest": {
[... 1415 lines cut ...]
    "dcPacStr": "kW",
    "psum": 0.014,
    "familyLoadPowerPec": "1",
    "psumCal": 0.014,
    "backup2PowerOriginal": 0.0,
    "psumStr": "kW",
    "psumCalStr": "kW",
    "bypassLoadPowerOriginal": 0.0,
    "reactivePowerStr": "Var",
    "apparentPowerStr": "VA"
  }
}
2025-05-20 22:33:52.047 WARNING (MainThread) [custom_components.solis_cloud_control.inverters.inverter_factory] Unknown inverter model '3306', fallback to generic hybrid inverter
2025-05-20 22:33:52.048 DEBUG (MainThread) [custom_components.solis_cloud_control.api.solis_api] API request '/v2/api/atReadBatch': {
  "inverterSn": "1033060233210147",
  "cids": ""
}
2025-05-20 22:33:52.091 DEBUG (MainThread) [custom_components.solis_cloud_control.api.solis_api] API response: {
  "success": true,
  "code": "I0000",
  "msg": "The necessary parameters are empty or invalid",
  "data": null
}

We appreciate you taking the time to provide this information. Adding support for new hardware can be complex, and your detailed input is crucial for success!

mkuthan, 2025-05-21:
Thanks for detailed issue, it helped me a lot!

Please let me know if the latest version, 2.2.2, is compatible with your inverter. In this release, the S6-EH3P10K-H inverter is treated as a "generic" hybrid inverter. Ensure that all entities function properly and display the correct settings. For instance, verify whether the maximum export power is scaled appropriately; some inverters measure export power in watts, while others use units of 100 watts.

closed 2025-05-23 by mkuthan

jesperkroghc, 2025-05-23:
It works now! Thanks a lot! VhJesper Krogh ChristensenMøllemoseparken 1853450 Allerød

Closed #32 as completed via #33.

PR #33 (2025-05-23): feat: add support for S6-EH3P(5-10)K-H inverter model
Closes #32