mkuthan/solis-cloud-control #37: Support Request: S5-EH1P(3-6)K-L-EXTENSION-10

pmocrob, 2025-06-02:
Thank you for your interest in expanding the capabilities of this integration! To help us add support for your new Solis inverter, please provide as much of the following information as possible.

**1. Inverter Model Name/Type:**
Please provide the exact model name/type of your Solis inverter (e.g., S6-EH3P12K02-NV-YD-L).

```
S5-EH1P3.6K-L-DC
```

**2. Inverter Model Identifier/Number:**
Please provide the model identifier/number of your Solis inverter (e.g., "3331").

```
3102
```

**3. Diagnostics Information:**
Download diagnostics from Home Assistant integration page. Please paste `inverter_info` and `coordinator_data` fields within the code block below.

```json
{
    "inverter_info": {
      "serial_number": "**REDACTED**",
      "model": "3102",
      "version": "3F003C",
      "machine": "S5-EH1P3.6K-L",
      "energy_storage_control": "23",
      "smart_support": "0",
      "generator_support": "0",
      "power": "3.6",
      "power_unit": "kW"
    },
    "coordinator_data": {
      "52": "190",
      "54": "190",
      "636": "35",
      "103": null,
      "5916": "0",
      "5946": "00:01-2300:5000",
      "5948": "0",
      "5928": "0",
      "5917": "0",
      "5949": "00:500-420:535",
      "5951": "0",
      "5929": "1",
      "5918": "0",
      "5952": "20:10-00:00",
      "5954": "60",
      "5930": "532",
      "5919": "0",
      "5955": "00:00-00:00",
      "5957": "0",
      "5931": "0",
      "5920": "0",
      "5958": "00:00-00:00",
      "5960": "0",
      "5932": "0",
      "5921": "0",
      "5961": "00:00-00:00",
      "5963": "0",
      "5933": "0",
      "5922": "0",
      "5964": "00:00-00:00",
      "5967": "0",
      "5965": "0",
      "5923": "0",
      "5968": "00:00-02:2450",
      "5971": "0",
      "5969": "0",
      "5924": "0",
      "5972": "00:00-00:145",
      "5975": "0",
      "5973": "112",
      "5925": "0",
      "5976": "00:00-145:00",
      "5979": "0",
      "5977": "0",
      "5926": "0",
      "5980": "00:00-00:92",
      "5983": "0",
      "5981": "92",
      "5927": "0",
      "5987": "00:00-00:00",
      "5986": "17.4",
      "5984": "0",
      "499": "0",
      "157": "80",
      "158": "20",
      "160": "10",
      "7229": "20",
      "7963": "100",
      "7224": "100",
      "7226": "100"
    }

```

**4. Additional Details:**
Please provide any other relevant information that might help us add support for this inverter. For example:

- What specific features or sensors are you hoping to see?
- Screenshots of the inverter's web interface or app with settings you would like to control
- Logs or error messages from the integration that you have encountered

I'd like to be able to easily configure some discharge times, but they appear to be unavailable.

[image]

We appreciate you taking the time to provide this information. Adding support for new hardware can be complex, and your detailed input is crucial for success!

mkuthan, 2025-06-07:
1. I will submit a new request to upgrade the inverter firmware.
2. Could you please check the current inverter settings, specifically “Storage Mode” and “Charge & Discharge Slots,” in the Solis app or website? Screenshots would be helpful.

I have noticed that the storage mode (cid: 636) is set to 35 (binary: 0b100011), which means the following flags are enabled:
 
- bit 0 - self-use
- bit 1 - time-of-use
- bit 5 - allow charging from the grid

Currently, the integration does not handle “bit 1 / time-of-use” because my inverter does not utilize this flag, even if I schedule charge or discharge slots. This functionality might differ between the S5 and S6 hybrid inverters.

uksysadmin, 2025-06-08:
[image]

[image]

[image]

[image]

mkuthan, 2025-06-08:
Sorry, but the Solis API response from your inverter regarding charge/discharge slots is garbage. See for yourself: this is an API specification snippet. Compare it with "coordinator_data" from the integration diagnostic output.

| Register ID | Description                     |
|-------------|---------------------------------|
| 5916        | Charge Time Slot 1 Switch      |
| 5917        | Charge Time Slot 2 Switch      |
| 5918        | Charge Time Slot 3 Switch      |
| 5919        | Charge Time Slot 4 Switch      |
| 5920        | Charge Time Slot 5 Switch      |
| 5921        | Charge Time Slot 6 Switch      |
| 5922        | Discharge Time Slot 1 Switch   |
| 5923        | Discharge Time Slot 2 Switch   |
| 5924        | Discharge Time Slot 3 Switch   |
| 5925        | Discharge Time Slot 4 Switch   |
| 5926        | Discharge Time Slot 5 Switch   |
| 5927        | Discharge Time Slot 6 Switch   |
| 5946        | Charge Time Slot 1             |
| 5928        | SOC1                           |
| 5947        | Volt 1                         |
| 5948        | Charge Current 1               |
| 5949        | Charge Time Slot 2             |
| 5929        | SOC2                           |
| 5950        | Volt 2                         |
| 5951        | Charge Current 2               |
| 5952        | Charge Time Slot 3             |
| 5930        | SOC3                           |
| 5953        | Volt 3                         |
| 5954        | Charge Current 3               |
| 5955        | Charge Time Slot 4             |
| 5931        | SOC4                           |
| 5956        | Volt 4                         |
| 5957        | Charge Current 4               |
| 5958        | Charge Time Slot 5             |
| 5932        | SOC5                           |
| 5959        | Volt 5                         |
| 5960        | Charge Current 5               |
| 5961        | Charge Time Slot 6             |
| 5933        | SOC6                           |
| 5962        | Volt 6                         |
| 5963        | Charge Current 6               |
| 5964        | Discharge Time Slot 1          |
| 5965        | SOC1                           |
| 5966        | Volt 1                         |
| 5967        | Discharge Current 1            |
| 5968        | Discharge Time Slot 2          |
| 5969        | SOC2                           |
| 5970        | Volt 2                         |
| 5971        | Discharge Current 2            |
| 5972        | Discharge Time Slot 3          |
| 5973        | SOC3                           |
| 5974        | Volt 3                         |
| 5975        | Discharge Current 3            |
| 5976        | Discharge Time Slot 4          |
| 5977        | SOC4                           |
| 5978        | Volt 4                         |
| 5979        | Discharge Current 4            |
| 5980        | Discharge Time Slot 5          |
| 5981        | SOC5                           |
| 5982        | Volt 5                         |
| 5983        | Discharge Current 5            |
| 5987        | Discharge Time Slot 6          |
| 5984        | SOC6                           |
| 5985        | Volt 6                         |
| 5986        | Discharge Current 6            |

Initially I thought your inverter was using the old 3-slot format with common charge and discharge current. It should look like this:

```json
{"cid":"103","value":"100,100,09:44,09:44,09:44,09:44,100,100,09:44,09:44,09:44,09:44,99,100,09:44,09:44,09:44,09:44"}
```

But API response for register "103" is empty. I'm afraid that only Solis support could help for this mess.

mkuthan, 2025-06-09:
It looks that your inverter still doesn't use TOU v2 (Time of Use v2) settings. There is a special flag under register "6798", when it is set to "43605" (or "0xAA55" in hex) inverter supports charge / discharge slots as expected. It's clearly visible that your inverter (or firmware) returns "0" for "6798" register, it also returns "0" and "00:00-00:00" for all charge/discharge slots related registers.

If you can capture the network traffic from the official Solis webpage used to control the inverter, by utilizing the browser's built-in developer console, we can proceed further. See: https://developer.chrome.com/docs/devtools/network/overview

The requests and responses are in plain JSON format with visible registers, such as the "cid" parameter, and contain detailed responses. To gather more insights, try configuring some charge/discharge slots and take note of the registers and their respective values.

uksysadmin, 2025-06-09:
I captured navigation through the menu to self-use, charge and discharge. Captured this below.
I guess the more interesting part is the setting of the new time, seen at the bottom.

The first few payload/response is navigation/reading of values.

However, since I was doing a search/replace to REDACT my serial number, I was intrigued, on successfully setting my new time, that it randomly presented someone else's inverter serial number in the output. That to me seems very odd.
I've removed a few numbers just in case this is something that could be exploited.

Work Mode
    Self-Use Mode
        {;} tree

Response:

{
    "success": true,
    "code": "0",
    "msg": "success",
    "data": [
        {
            "id": "466",
            "widgetType": 5,
            "inverterModelId": "2",
            "registerAddr": 43110,
            "registerNumber": 1,
            "buttonShow": 0,
            "buttonControlShow": 0,
            "menu": "[{\"name\":\"Enable\",\"value\":\"1\"},{\"name\":\"Disable\",\"value\":\"0\"}]",
            "pid": "96",
            "name": "Self-Use Mode Switch",
            "notice": "Self-Use, Feed in priority mode ,Peak-shaving and Off-grid mode cannot be turned off at the same time. After turning on other modes, Self-Use will be turned off automatically.",
            "content": "The switch is to enable/disable the self-use mode",
            "passWordCheck": false,
            "passWordCheckType": 0,
            "needRefresh": 1,
            "freshByRules": 0,
            "isPageShow": 0,
            "noticeIsOutShow": false,
            "isJiChu": 0,
            "on": 1,
            "off": 0
        },
        {
            "id": "543",
            "widgetType": 5,
            "inverterModelId": "2",
            "registerAddr": 43110,
            "registerNumber": 1,
            "buttonShow": 0,
            "buttonControlShow": 0,
            "menu": "[{\"name\":\"Enable\",\"value\":\"1\"},{\"name\":\"Disable\",\"value\":\"0\"}]",
            "pid": "96",
            "name": "Time of Use Switch",
            "content": "The switch is to enable/disable the Time of use setting to allow user to set a customized charging/discharging pattern",
            "passWordCheck": false,
            "passWordCheckType": 0,
            "needRefresh": 1,
            "freshByRules": 0,
            "isPageShow": 0,
            "noticeIsOutShow": false,
            "isJiChu": 0,
            "on": 1,
            "off": 0
        },
        {
            "id": "103",
            "widgetType": 9,
            "inverterModelId": "2",
            "registerAddr": 43141,
            "registerNumber": 30,
[... 532 lines cut ...]
localTimeZone
: 
1
value
: 
"20,25,04:30-05:30,05:30-04:30,0,0,00:00-00:00,00:00-00:00,0,0,00:00-00:00,00:00-00:00"
yuanzhi
: 
"20,25,03:30-05:30,05:30-03:30,0,0,00:00-00:00,00:00-00:00,0,0,00:00-00:00,00:00-00:00"

Response:

{
    "success": true,
    "code": "0",
    "data": [
        {
            "msg": "Send Success !REDACTED    Time: 1749468414<br>NO: 423<br>SN: 649029073XXXX664435<br>Result: 3<br>Inverter1: <br>send: <br>recv: 0110A885001E7188<br>",
            "code": 0,
            "recv": "0110A885001E7188",
            "needRefresh": 0,
            "command": "AT+TEST=GIN485:01 10 a8 85 00 1e 3c 00 c8 00 fa 00 04 00 1e 00 05 00 1e 00 05 00 1e 00 04 00 1e 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 75 dd"
        }
    ]
}

uksysadmin, 2025-06-14:
Thanks it's getting closer.  This now presents the correct string as you stated:

[image]

But I don't have a way of updating the field. When I go to edit the values, I get an error:

[image]

uksysadmin, 2025-06-14:
I've just used Developer Tools to see if I can spot any formatting errors and I can't

This is an equivalent set in Solis Cloud:

{"cid":"103","inverterId":"REDACTED","value":"20,25,03:30-05:30,05:30-03:30,0,0,00:00-00:00,00:00-00:00,0,0,00:00-00:00,00:00-00:00","yuanzhi":"20,25,04:30-05:30,05:30-23:30,0,0,23:30-01:30,01:30-04:30,0,0,00:00-00:00,00:00-00:00","localTime":1749919332452,"localTimeZone":1,"language":"2"}

mkuthan, 2025-06-14:
Done :)
https://github.com/mkuthan/solis-cloud-control/releases/tag/v2.7.0-beta.2

mkuthan, 2025-06-16:
I replaced single text entity with separate entities for time slots and charge/discharge currents.
Could you check https://github.com/mkuthan/solis-cloud-control/releases/tag/v2.7.0-beta.3 please?

uksysadmin, 2025-06-17:
I deployed beta3. It shows the values in their own entry points now.

[image]

I can set a charge, but I can't set a discharge - I get invalid entry.
And it appears the discharge isn't showing correctly.

I set a charge/discharge time in Solis Cloud and it doesn't show in HASS.

[image]

[image]

mkuthan, 2025-06-17:
Most likely I found the bug, please check: https://github.com/mkuthan/solis-cloud-control/releases/tag/v2.7.0-beta.4 :)

uksysadmin, 2025-06-18:
I can confirm this has now fixed any outstanding issues I had related to my device. I can set modes and charge/discharge slots. Thank you!

mkuthan, 2025-06-18:
Thanks for cooperation! Implemented in v2.7.0

closed 2025-06-18 by mkuthan