mkuthan/solis-cloud-control #39: Support Request: S6-GR1P5K (0205)

Bodmoor, 2025-06-07:
Thank you for your interest in expanding the capabilities of this integration! To help us add support for your new Solis inverter, please provide as much of the following information as possible.

**1. Inverter Model Name/Type:**
Please provide the exact model name/type of your Solis inverter (e.g., S6-EH3P12K02-NV-YD-L).

```
S6-GR1P5K (0205)
```

**2. Inverter Model Identifier/Number:**
Please provide the model identifier/number of your Solis inverter (e.g., "3331").

```
0205
```

**3. Diagnostics Information:**
Download diagnostics from Home Assistant integration page. Please paste `inverter_info` and `coordinator_data` fields within the code block below.

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2025.5.3",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.13.3",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Amsterdam",
    "os_name": "Linux",
    "os_version": "6.6.74-haos-raspi",
    "supervisor": "2025.05.3",
    "host_os": "Home Assistant OS 15.2",
    "docker_version": "28.0.4",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "solis_cloud_control": {
      "documentation": "https://github.com/mkuthan/solis-cloud-control",
      "version": "2.5.0",
      "requirements": []
    },
    "smartlife": {
      "documentation": "https://www.home-assistant.io/integrations/smartlife",
      "version": "0.1.0",
      "requirements": [
        "tuya-device-sharing-sdk==0.2.0"
      ]
    },
    "pyscript": {
      "documentation": "https://github.com/custom-components/pyscript",
      "version": "1.6.4",
      "requirements": [
        "croniter==6.0.0",
        "watchdog==6.0.0"
      ]
    },
    "zonneplan_one": {
      "documentation": "https://github.com/fsaris/home-assistant-zonneplan-one",
      "version": "2025.6.0",
      "requirements": []
    },
    "multiscrape": {
      "documentation": "https://github.com/danieldotnl/ha-multiscrape",
      "version": "8.0.2",
      "requirements": [
        "lxml>=4.9.1",
        "beautifulsoup4>=4.12.2"
      ]
    },
    "frank_energie": {
      "documentation": "https://github.com/bajansen/home-assistant-frank_energie",
      "version": "0.0.1",
      "requirements": []
    },
    "degree_days": {
      "documentation": "https://www.home-assistant.io/integrations/degree_days",
      "version": "0.8.0",
      "requirements": []
    },
    "hacs": {
      "documentation": "https://hacs.xyz/docs/use/",
      "version": "2.0.5",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    }
  },
  "integration_manifest": {
    "domain": "solis_cloud_control",
    "name": "Solis Cloud Control Integration",
    "codeowners": [
      "mkuthan"
    ],
    "config_flow": true,
    "documentation": "https://github.com/mkuthan/solis-cloud-control",
    "iot_class": "cloud_polling",
    "issue_tracker": "https://github.com/mkuthan/solis-cloud-control/issues",
    "version": "2.5.0",
    "is_built_in": false,
    "overwrites_built_in": false
  },
  "setup_times": {
    "null": {
      "setup": 2.7148984372615814e-05
    },
    "01JX4J82230CFJ46HJMXPBMJ85": {
      "wait_import_platforms": -0.055465943180024624,
      "wait_base_component": -0.0008832849562168121,
      "config_entry_setup": 11.283401277847588
    }
  },
  "data": {
    "inverter_info": {
      "serial_number": "**REDACTED**",
      "model": "0205",
      "version": "780035",
      "machine": "S6-GR1P5K",
      "energy_storage_control": "0",
      "smart_support": "0",
      "generator_support": "0",
      "collector_model": "WIFI",
      "power": "5.0",
      "power_unit": "kW"
    },
    "coordinator_data": {
      "48": "222",
      "53": "222",
      "15": "100"
    }
  }
}
```

**4. Additional Details:**
Please provide any other relevant information that might help us add support for this inverter. For example:

- What specific features or sensors are you hoping to see?
- Screenshots of the inverter's web interface or app with settings you would like to control
- Logs or error messages from the integration that you have encountered

I cannot see any error and cannot see where the logs are stored, but the button for on/off resets to on when turning off. Also, the power setting percentage slider does not work, it also reverts. And since the power setting can go from 0 to 110, the slider might not be sufficient (since it only goes to 100).

If you can point me to the logs I will include them.

We appreciate you taking the time to provide this information. Adding support for new hardware can be complex, and your detailed input is crucial for success!

Bodmoor, 2025-06-07:
I found my log:

```
custom_components.solis_cloud_control.switch] Turning off inverter
2025-06-07 11:14:16.090 DEBUG (MainThread) [custom_components.solis_cloud_control.api.solis_api] API request '/v2/api/control': {
  "inverterSn": "1802050226180085",
  "cid": 53,
  "value": "222"
}
2025-06-07 11:14:16.409 DEBUG (MainThread) [custom_components.solis_cloud_control.api.solis_api] API response: {
  "code": "0",
  "data": [
    {
      "msg": "Send Success !1802050226180085    Time: 1749291140<br>NO: 788<br>SN: 6490290864817984189<br>Result: 0<br>Inverter1: <br>send: <br>recv: 01060BBE00DE6B92<br>",
      "code": 0,
      "recv": "01060BBE00DE6B92",
      "command": "AT+TEST=GIN485:01 06 0b be 00 de 6b 92"
    }
  ],
  "time": "1749287656167"
}
2025-06-07 11:14:16.410 DEBUG (MainThread) [custom_components.solis_cloud_control.api.solis_api] API request '/v2/api/atReadBatch': {
  "inverterSn": "1802050226180085",
  "cids": "48,53,15"
}
2025-06-07 11:14:16.903 DEBUG (MainThread) [custom_components.solis_cloud_control.api.solis_api] API response: {
  "success": true,
  "code": "0",
  "msg": "success",
  "data": [
    [
      {
        "msg": "222",
        "yuanzhi": "222",
        "sysCommand": {
          "id": "48",
          "createDate": 1656582722000,
          "updateDate": 1745462729000,
          "remarks": "\u5e76\u7f51,\u5f00",
          "deleteFlag": 0,
          "inverterModelId": "1",
          "widgetType": 0,
          "type": "device_switch_on",
          "functionCode": 6,
          "registerAddr": 3006,
          "functionCodeRead": 3,
          "registerAddrRead": 3006,
          "registerNumber": 1,
          "value": "1",
          "yzRead": "1107,r_kgjsn",
          "yzWrite": "1107,s_kgjsn",
          "yzRead0505": "1100,r_kgjsn",
          "yzWrite0505": "1100,s_kgjsn",
          "productModel": "1,2101,2102",
          "level": 0,
          "name1": "\u5f00",
          "name2": "on",
          "name3": "ENCENDIDO",
          "name5": "EIN",
          "name9": "LIGADO",
          "name10": "ACCESO",
          "isJiChu": 2,
          "menu1": "190",
          "menu2": "190",
          "needRead": 0,
          "bit": -1,
          "unbit": "-1",
          "othersOnbit": "-1",
          "xiaoshu": -1,
          "auth": 0,
[... 372 lines cut ...]
          "passWordCheckType": 0,
          "needRefresh": 0,
          "freshByRules": 0,
          "idStr": "15"
        },
        "command": "AT+TEST=GIN485:01 03 0b be 00 2e a7 d6",
        "cid": "15"
      }
    ]
  ]
}
2025-06-07 11:17:16.962 DEBUG (MainThread) [custom_components.solis_cloud_control.coordinator] Data read from API: {48: '222', 53: '222', 15: '100'}
2025-06-07 11:17:16.962 DEBUG (MainThread) [custom_components.solis_cloud_control.coordinator] Finished fetching Solis Cloud Control data in 0.493 seconds (success: True)
2025-06-07 11:17:30.269 INFO (MainThread) [homeassistant.components.automation.scrape_solar] Scrape solar: Running automation actions
2025-06-07 11:17:30.269 INFO (MainThread) [homeassistant.components.automation.scrape_solar] Scrape solar: Executing step call service
2025-06-07 11:17:30.269 INFO (MainThread) [custom_components.multiscrape.service] Multiscrape triggered by service: <ServiceCall multiscrape.trigger_solis_scraper (c:01JX4TWM8WZFFCVD240QBW70GV)>
2025-06-07 11:18:00.269 INFO (MainThread) [homeassistant.components.automation.scrape_solar] Scrape solar: Running automation actions
2025-06-07 11:18:00.270 INFO (MainThread) [homeassistant.components.automation.scrape_solar] Scrape solar: Executing step call service
2025-06-07 11:18:00.270 INFO (MainThread) [custom_components.multiscrape.service] Multiscrape triggered by service: <ServiceCall multiscrape.trigger_solis_scraper (c:01JX4TXHJDK61MJ8W9QQPGQK5V)>
2025-06-07 11:18:30.269 INFO (MainThread) [homeassistant.components.automation.scrape_solar] Scrape solar: Running automation actions
2025-06-07 11:18:30.269 INFO (MainThread) [homeassistant.components.automation.scrape_solar] Scrape solar: Executing step call service
2025-06-07 11:18:30.269 INFO (MainThread) [custom_components.multiscrape.service] Multiscrape triggered by service: <ServiceCall multiscrape.trigger_solis_scraper (c:01JX4TYEVW0F3SDMCRZZ54WC2P)>
2025-06-07 11:18:44.739 ERROR (MainThread) [homeassistant.components.emulated_hue.hue_api] Entity not found: light.smoke_detector_basic

```

mkuthan, 2025-06-07:
@Bodmoor - could you check https://github.com/mkuthan/solis-cloud-control/releases/tag/v2.6.0-alpha.1, please? OnOff switch should show current inverter state.

Thanks!

closed 2025-06-07 by Bodmoor

Bodmoor, 2025-06-08:
It is indeed the S3-WiFi-ST, 2AWE8 is the FCC Id prefix if it matters.

The slider works great, lower settings like 46 are received well, and resetting to 110 also works.

Thanks!