mkuthan/solis-cloud-control #51: New Solis Inverter Support Request S6-EH3P50K-H - disable export

valev1, 2025-07-06:
Thank you for your interest in expanding the capabilities of this integration! To help us add support for your new Solis inverter, please provide as much of the following information as possible.

**1. Inverter Model Name/Type:**
Please provide the exact model name/type of your Solis inverter (e.g., S6-EH3P12K02-NV-YD-L).

```
S6-EH3P50K-H  Hybrid 50kW
```

**2. Inverter Model Identifier/Number:**
Please provide the model identifier/number of your Solis inverter (e.g., "3331").

```
5305
```

**3. Diagnostics Information:**
Download diagnostics from Home Assistant integration page. Please paste `inverter_info` and `coordinator_data` fields within the code block below.

```json
"data": {
    "inverter_info": {
      "serial_number": "**REDACTED**",
      "model": "5305",
      "version": "030003",
      "machine": "S6-EH3P50K-H",
      "energy_storage_control": "1",
      "smart_support": "1",
      "generator_support": "1",
      "collector_model": "WL",
      "power": "50.0",
      "power_unit": "kW",
      "parallel_number": "1.0",
      "parallel_battery": "1",
      "tou_v2_mode": "43605"
    },
    "coordinator_data": {
      "52": "190",
      "54": "190",
      "636": "1",
      "5916": "0",
      "5946": "00:00-00:00",
      "5948": "0",
      "5928": "25",
      "5917": "0",
      "5949": "00:00-00:00",
      "5951": "0",
      "5929": "30",
      "5918": "0",
      "5952": "00:00-00:00",
      "5954": "0",
      "5930": "30",
      "5919": "0",
      "5955": "00:00-00:00",
      "5957": "0",
      "5931": "30",
      "5920": "0",
      "5958": "00:00-00:00",
      "5960": "0",
      "5932": "50",
      "5921": "0",
      "5961": "11:00-17:00",
      "5963": "140",
      "5933": "98",
      "5922": "0",
      "5964": "08:00-08:59",
      "5967": "50",
      "5965": "18",
      "5923": "0",
      "5968": "07:00-08:00",
      "5971": "50",
      "5969": "20",
      "5924": "0",
      "5972": "06:15-07:00",
      "5975": "40",
      "5973": "20",
      "5925": "0",
      "5976": "18:00-19:00",
      "5979": "60",
      "5977": "44",
      "5926": "1",
      "5980": "19:00-20:00",
      "5983": "60",
      "5981": "35",
      "5927": "1",
      "5987": "20:00-23:59",
      "5986": "60",
      "5984": "28",
      "376": "52",
      "499": "250",
      "157": "12",
      "158": "15",
      "160": "10",
      "7229": "17",
      "7963": "95",
      "7224": "70",
      "7226": "70"
```

**4. Additional Details:**
Please provide any other relevant information that might help us add support for this inverter. For example:
1. I need to be able to disable 'EXPORT to grid', this option is only available when storage mode is set to "self use". I made screenshot of invertor control page

- What specific features or sensors are you hoping to see?
- Screenshots of the inverter's web interface or app with settings you would like to control
- Logs or error messages from the integration that you have encountered

valev1, 2025-07-07:
> Could you check
> https://github.com/mkuthan/solis-cloud-control/releases/tag/v2.10.0-alpha.1?
>
> I implemented simple switch without any dependencies to the current
> inverter storage mode. Feel free to test what happens when you
> disable/enable export in feed-in-priority mode :P
>
> I have one more question. Does the integration provide the correct max
> export power for your device? Based on diagnostic info, max export
> power is 52W but perhaps it should be 5200W or so:
>
> ```
> "376": "52"
> ```

Hello
"376": "52"

This setting is set in my inverter as a percentage ot total power.
50kw *52% = 26kw limit

Currently need to limit max output power to this level.

I attach screenshot

Regards

mkuthan, 2025-07-08:
Thanks for explanation, it's clear. 

Have you tested [v2.10.0-alpha.1](https://github.com/mkuthan/solis-cloud-control/releases/tag/v2.10.0-alpha.1)? Are you able to toggle "allow export" switch? If it works I will release the final version.

PR #53 (2025-07-10): feat: Add support for Allow Export functionality in inverter integration
Closes #51
closed 2025-07-10 by mkuthan