mkuthan/solis-cloud-control #52: Support Request: Solis RHI-3P10K-HVES-5G

Creedy18, 2025-07-07:
Support Request: Solis RHI-3P10K-HVES-5G

I used the Solis app to set battery charging times, i.e i am on Octopus Energy and was charging the batteries 23.30 to 05.30 daily.

Then installed Solis Cloud Control Integration and Solis Modbus Integration to HA on my PC.

Since installing Solis Cloud and Solis Modbus on HA my inverter has stopped charging.

Batteries charge normally from PV with no issues.

I have now removed Solis Cloud Control Integration and Solis Modbus Integration from HA and the inverter is still not charging from the grid.

I have no automations set in HA as yet, its all read only currently.

Solis technical support and my installers are baffled....

Any help greatly received...thanks in advance

Creedy18, 2025-07-07:
Found it, self use is enabled and allow grid charging is enabled

skrachs, 2025-07-07:
And charging times are all set and enabled?

Creedy18, 2025-07-07:
Just found an EMS function on the overview screen, it has two options Revenue Maximisation and Timed Plans and they are both unticked, is that correct ?

skrachs, 2025-07-07:
I don't have that.
That is the new Solis AI function.
You can play with them.
Sounds like the Revenue Maximisation could be an automated solution and timed plans a manual with fixed charge and discharge times.

skrachs, 2025-07-07:
Maybe the installer doesn't know it yet.
It's a completely new feature from Solis

You can ask Solis Support
https://solis-service.solisinverters.com/en/support/tickets

I believe it is just a coincidence that it stopped working when you installed this integration.

Or maybe here you can find something useful
https://www.solisinverters.com/global/conews/Solis_LaunchesSmart_Charge%26Discharge_Functionality_gl.html

https://docs.google.com/presentation/d/1h-neCAQY3rfGjt87QkNSo3xS-r0L8e3AQOqu80ZyJyY/edit?slide=id.p1#slide=id.p1

mkuthan, 2025-07-07:
Have you managed to enable discharging to grid / charging from grid? Could you share your EMS settings and time slots settings, please?

Creedy18, 2025-07-08:
[image]

Creedy18, 2025-07-08:
[image]

mjatkey, 2025-07-10:
HI Marcin, I've mailed Solis to see if the issues with my inverters are connected to this rollout, maybe its half applied although I thought you had to sign up for it and was initially aimed at UK market? Thanks