mkuthan/solis-cloud-control #56: Support Request: S6-GR1P3K-M & S6-GR1P1K-M

jeroenschouwenberg, 2025-07-12:
Thank you for the work on this amazing integration! I have been using it successfully for some months up to and including v2.8.0. When I update to a newer version (>=v2.9.0) the controls become unavailable and the log shows `API operation failed: Datalogger returns data abnormally. Please try again later (API response code: B0600)`. My inverter models are not officially supported yet, so hereby the support request.

**1. Inverter Model Name/Type & Identifier/Number:**

```
| Type        | ID   |
| ----------- | ---- |
| S6-GR1P3K-M | 0105 |
| S6-GR1P1K-M | 0101 |
```

**2. Diagnostics Information:**
Download diagnostics from Home Assistant integration page. Please paste `inverter_info` and `coordinator_data` fields within the code block below.

- S6-GR1P3K-M
```json
"data": {
  "inverter_info": {
    "serial_number": "**REDACTED**",
    "model": "0105",
    "version": "780036",
    "machine": "S6-GR1P3K-M",
    "energy_storage_control": "0",
    "smart_support": "0",
    "generator_support": "0",
    "collector_model": "WIFI",
    "power": "3.0",
    "power_unit": "kW"
  },
  "coordinator_data": {
    "48": "190",
    "53": "190",
    "15": "29"
  }
}
```

- S6-GR1P1K-M
```json
"data": {
  "inverter_info": {
    "serial_number": "**REDACTED**",
    "model": "0101",
    "version": "800037",
    "machine": "S6-GR1P1K-M",
    "energy_storage_control": "0",
    "smart_support": "0",
    "generator_support": "0",
    "collector_model": "WIFI",
    "power": "1.0",
    "power_unit": "kW"
  },
  "coordinator_data": {
    "48": "190",
    "53": "190",
    "15": "29"
  }
}
```

**3. Additional Details:**

- v2.8.0 log (despite some timeouts, the integration ends up working fine):
```log
2025-07-12 13:18:55.986 WARNING (MainThread) [custom_components.solis_cloud_control.inverters.inverter_factory] Unknown inverter model '0101', fallback to generic string inverter
2025-07-12 13:18:55.989 WARNING (MainThread) [custom_components.solis_cloud_control.inverters.inverter_factory] Unknown inverter model '0105', fallback to generic string inverter
2025-07-12 13:18:56.146 WARNING (ImportExecutor_0) [py.warnings] /usr/local/lib/python3.13/site-packages/google/__init__.py:2: UserWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html. The pkg_resources package is slated for removal as early as 2025-11-30. Refrain from using this package or pin to Setuptools<81.
  __import__('pkg_resources').declare_namespace(__name__)

2025-07-12 13:19:04.145 WARNING (MainThread) [custom_components.solis_cloud_control.api.solis_api] Retrying due to error: API operation failed: Sending failure ,the current datalogger is offline or disconnected (API response code: B0115) (attempt 1/5)
[... 115 lines cut ...]
  File "/config/custom_components/solis_cloud_control/__init__.py", line 28, in async_setup_entry
    inverter_info = await create_inverter_info(api_client, inverter_sn)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solis_cloud_control/inverters/inverter_factory.py", line 25, in create_inverter_info
    tou_v2_mode = await api_client.read(inverter_sn, 6798)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solis_cloud_control/api/solis_api.py", line 77, in read
    return await self._with_retry(read_operation, retry_count, retry_delay)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solis_cloud_control/api/solis_api.py", line 251, in _with_retry
    raise err
  File "/config/custom_components/solis_cloud_control/api/solis_api.py", line 244, in _with_retry
    return await operation_closure()
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solis_cloud_control/api/solis_api.py", line 67, in read_operation
    data = await self._execute_request(self._READ_ENDPOINT, payload)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solis_cloud_control/api/solis_api.py", line 226, in _execute_request
    raise SolisCloudControlApiError(f"API operation failed: {error_msg}", response_code=code)
custom_components.solis_cloud_control.api.solis_api.SolisCloudControlApiError: API operation failed: Datalogger returns data abnormally. Please try again later (API response code: B0600)
2025-07-12 14:05:15.408 ERROR (MainThread) [homeassistant.components.automation.solar_panels_sc_3] Solar panels - Self-consumption mode: Error executing script. Error for call_service at pos 2: Error rendering data template: ValueError: Template error: int got invalid input 'unavailable' when rendering template '{{ states('number.inverter_control_**REDACTED**_power_limit') | int }}' but no default was specified
2025-07-12 14:05:15.409 ERROR (MainThread) [homeassistant.components.automation.solar_panels_sc_3] Error while executing automation automation.solar_panels_sc_3: Error rendering data template: ValueError: Template error: int got invalid input 'unavailable' when rendering template '{{ states('number.inverter_control_**REDACTED**_power_limit') | int }}' but no default was specified
2025-07-12 14:05:30.403 ERROR (MainThread) [homeassistant.components.automation.solar_panels_sc_3] Solar panels - Self-consumption mode: Error executing script. Error for call_service at pos 2: Error rendering data template: ValueError: Template error: int got invalid input 'unavailable' when rendering template '{{ states('number.inverter_control_**REDACTED**_power_limit') | int }}' but no default was specified
2025-07-12 14:05:30.404 ERROR (MainThread) [homeassistant.components.automation.solar_panels_sc_3] Error while executing automation automation.solar_panels_sc_3: Error rendering data template: ValueError: Template error: int got invalid input 'unavailable' when rendering template '{{ states('number.inverter_control_**REDACTED**_power_limit') | int }}' but no default was specified
```

mkuthan, 2025-07-12:
Thank you for all details. The changes between 2.8.x and 2.9.x are for hybrid inverters only. It seems to me that I didn’t change anything that could introduce a regression for string inverters. I will take a look. Apologies for the inconvenience, but without access to all inverters, I need to perform testing in the production environment with end users.

mkuthan, 2025-07-20:
I found the bug, could you check v2.10.2 please?

jeroenschouwenberg, 2025-07-24:
Hi, thanks for the update! I just tested v2.10.2 and it works perfectly again! The log is also completely clean.
Thanks again for making this integration and the fast support!!

closed 2025-07-24 by jeroenschouwenberg