mkuthan/solis-cloud-control #57: Support Request: [3P10K-4G/Type]

MijnSpam, 2025-07-13:
Thank you for your interest in expanding the capabilities of this integration! To help us add support for your new Solis inverter, please provide as much of the following information as possible.

**1. Inverter Model Name/Type:**
Please provide the exact model name/type of your Solis inverter (e.g., S6-EH3P12K02-NV-YD-L).

```
Solis-3P10K-4G
```

**2. Inverter Model Identifier/Number:**

```
app says model; 7A
Inverter 1107A220A230004
```

**3. Diagnostics Information:**
Download diagnostics from Home Assistant integration page. Please paste `inverter_info` and `coordinator_data` fields within the code block below.

```json
Is setting on debug logging needed first?

```

**4. Additional Details:**
Please provide any other relevant information that might help us add support for this inverter. For example:
I can enter my key and secret during setup, before device is added you wait on this screen (which does continu)

[image]
next the device is added to an area.
The inverter section goes to initialise status and after a while you get the following error:

[image]

mkuthan, 2025-07-13:
What data logger model do you have?

See:
https://github.com/mkuthan/solis-cloud-control/?tab=readme-ov-file#what-if-the-integration-reports-b0115-error-during-initialization

MijnSpam, 2025-07-13:
This info is from the datalogger interface: 
Firmware version (main)001E
Firmware version (slave)002E
Inverter model007A

Device information:
Firmware version ME_0C_0501_5.08

I don't see anything similar to the mentioned examples

mkuthan, 2025-07-13:
Thanks for response but I asked for data logger model, for example S2-WL-ST, S3-WiFi-ST, DLS-W, DLS-L. etc.
https://usservice.solisinverters.com/support/solutions/articles/73000593957-solis-data-loggers-explained

Only selected data loggers support Solis Cloud Control API in a way compatible with this integration.