mkuthan/solis-cloud-control #58: Support Request: [Solis S5-GR3P8K]

Hein1965, 2025-07-19:
To help us add support for your new Solis inverter, please provide as much of the following information as possible.

**1. Inverter Model Name/Type:**
Please provide the exact model name/type of your Solis inverter (e.g., S6-EH3P12K02-NV-YD-L).

```
[Solis S5-GR3P8K]
```

**2. Inverter Model Identifier/Number:**
Please provide the model identifier/number of your Solis inverter (e.g., "3331").

```
[0504]
```

**3. Diagnostics Information:**
Download diagnostics from Home Assistant integration page. Please paste `inverter_info` and `coordinator_data` fields within the code block below.

```json
[  "data": {
    "inverter_info": {
      "serial_number": "**REDACTED**",
      "model": "0504",
      "version": "740035",
      "machine": "S5-GR3P8K",
      "energy_storage_control": "0",
      "smart_support": "0",
      "generator_support": "0",
      "collector_model": "WIFI",
      "power": "8.0",
      "power_unit": "kW"
    },
    "coordinator_data": {
      "48": "190",
      "53": "190",
      "15": "110"]
```

**4. Additional Details:**
Please provide any other relevant information that might help us add support for this inverter. For example:

My data logger is Wifi-based, with Model no. S3-WIFI-ST

Version 2.8.0 is working fine for me..!.

As of version 2.9 the integration still discovers my inverter with no issues (after inputting API key and token). But when I press submit, the integration takes some time trying to install and eventually responds with ‘Failed to set up’.

Logs for version 2.9.0 (and newer):
Logger: custom_components.solis_cloud_control.api.solis_api
Source: custom_components/solis_cloud_control/api/solis_api.py:248
integration: Solis Cloud Control Integration ([documentation](https://github.com/mkuthan/solis-cloud-control), [issues](https://github.com/mkuthan/solis-cloud-control/issues))
First occurred: 5:47:26 PM (2 occurrences)
Last logged: 5:47:32 PM
•	Retrying due to error: API operation failed: Datalogger returns data abnormally. Please try again later (API response code: B0600) (attempt 1/2)
•	Retrying due to error: API operation failed: Datalogger returns data abnormally. Please try again later (API response code: B0600) (attempt 2/2)

mkuthan, 2025-07-20:
I found the bug, could you check v2.10.2 please?

Hein1965, 2025-07-20:
Hi Marcin,
I have tested v2.10.2 and it seems to work fine for me.
Suppose you can add Solis S5-GR3PK8 to the list of supported devices (refer to attached)
Thank you so much for the good work..!

[image]