mkuthan/solis-cloud-control #59: Support Request: [RAI-3K-48ES-5G]

ElGringo33, 2025-07-21:
To help us add support for your new Solis inverter, please provide as much of the following information as possible.

**1. Inverter Model Name/Type:**
Please provide the exact model name/type of your Solis inverter (e.g., S6-EH3P12K02-NV-YD-L).

```
[SOLIS RAI-3K-48ES-5G]
```

**2. Inverter Model Identifier/Number:**
Please provide the model identifier/number of your Solis inverter (e.g., "3331").

```
[Inverter 010262226190101]
```

**3. Diagnostics Information:**
Download diagnostics from Home Assistant integration page. Please paste `inverter_info` and `coordinator_data` fields within the code block below.

```json
[{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2025.6.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.13.3",
    "docker": true,
    "arch": "aarch64",
    "timezone": "Europe/Paris",
    "os_name": "Linux",
    "os_version": "6.6.74-haos-raspi",
    "supervisor": "2025.05.5",
    "host_os": "Home Assistant OS 15.2",
    "docker_version": "28.0.4",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "tarif_edf": {
      "documentation": "https://github.com/delphiki/hass-tarif-edf",
      "version": "2.1.1",
      "requirements": []
    },
    "hacs": {
      "documentation": "https://hacs.xyz/docs/use/",
      "version": "2.0.5",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "solis_cloud_control": {
      "documentation": "https://github.com/mkuthan/solis-cloud-control",
      "version": "2.10.2",
      "requirements": []
    },
    "solis": {
      "documentation": "https://github.com/hultenvp/solis-sensor/",
      "version": "3.12.0",
      "requirements": [
        "aiofiles>=23.1.0,<25.0.0"
      ]
    },
    "solcast_solar": {
      "documentation": "https://github.com/BJReplay/ha-solcast-solar",
      "version": "v4.3.5",
      "requirements": [
        "aiohttp>=3.8.5",
        "aiofiles>=23.2.0",
        "datetime>=4.3"
      ]
    },
    "scheduler": {
      "documentation": "https://github.com/nielsfaber/scheduler-component",
      "version": "v0.0.0",
      "requirements": []
    },
    "nodered": {
      "documentation": "https://zachowj.github.io/node-red-contrib-home-assistant-websocket/guide/custom_integration/",
      "version": "4.1.2",
      "requirements": []
    },
    "myenergi": {
      "documentation": "https://github.com/cjne/ha-myenergi",
      "version": "0.0.30",
      "requirements": [
        "pymyenergi==0.2.3"
      ]
    }
  },
  "integration_manifest": {
    "domain": "solis_cloud_control",
    "name": "Solis Cloud Control Integration",
    "codeowners": [
      "mkuthan"
    ],
    "config_flow": true,
    "documentation": "https://github.com/mkuthan/solis-cloud-control",
    "iot_class": "cloud_polling",
    "issue_tracker": "https://github.com/mkuthan/solis-cloud-control/issues",
    "version": "2.10.2",
    "is_built_in": false,
    "overwrites_built_in": false
  },
  "setup_times": {
    "null": {
      "setup": 5.0332979299128056e-05
    },
    "01K0MX545B9ECXRW9AJ20HGW3W": {
      "wait_import_platforms": -0.031823377939872444,
      "wait_base_component": -0.0017317039892077446,
      "config_entry_setup": 19.1630035310518
    }
  },
  "data": {
    "inverter_info": {
      "serial_number": "**REDACTED**",
      "model": "26",
      "version": "18001A",
      "machine": "RAI-3K-48ES-5G",
      "energy_storage_control": "23",
      "smart_support": "0",
      "generator_support": "0",
      "collector_model": "WL",
      "power": "3.0",
      "power_unit": "kW",
      "parallel_number": "0.0",
      "parallel_battery": "0",
      "tou_v2_mode": "0"
    },
    "coordinator_data": {
      "52": "190",
      "54": "190",
      "636": "3",
      "376": "110",
      "499": "0",
      "6962": "0",
      "157": "80",
      "158": "20",
      "160": "10",
      "7229": "20",
      "7963": "100",
      "7224": "62.5",
      "7226": "62.5",
      "103": "20,50,00:00-06:25,00:00-00:00,0,0,00:00-00:00,00:00-00:00,0,0,00:00-00:00,00:00-00:00"
    }
  }
}]
```

**4. Additional Details:**
Please provide any other relevant information that might help us add support for this inverter. For example:

Hi, first of all thank you for the great integration!

I'm encountering an issue with the ON/OFF control of the inverter. While some commands are working correctly — for example:

    Setting the charging current

    Modifying the Time of Use – Slot 1 schedule

…the ON/OFF control (switch.turn_off) does not work.
When I trigger this command, I get the following error:

switch/turn_off. Control failed: Send failed! Lack of iot platform three elements (API response code: 1)

It seems like the API call is missing key elements required by the Solis IoT platform, but the rest of the integration appears correctly configured, as other features do work.

Is this a limitation of the API, or is there something I might have missed in the configuration?

Thanks in advance for your help!

mkuthan, 2025-07-21:
The on/off is hardly documented in Solis Cloud Control API "specification". I have never observed errors like "Lack of iot platform three elements" before so I need more details from you :)

Could you inspect network traffic from official Solis web application for on/off operations? I'm particularly interested in requests to "atRead" and "control" endpoints when you disable/enable your inverter. On/off control requests should have "cid" parameter set to 52 for on, 54 for off. Based on the requests from the official Solis web page I should be able to reverse-engineer the protocol used by your inverter.

See: https://developer.chrome.com/docs/devtools/network

Based on enclosed diagnostic info, I see that your inverter correctly reports its on state (value 190 for "cids" 52 and 54). But I don't know why on/off can't be controlled by the integration.

```json
"coordinator_data": {
      "52": "190",
      "54": "190"
}
,,,

ElGringo33, 2025-07-21:
Hi,

As requested, I’ve inspected the network traffic while using the official SolisCloud web interface to control my inverter. Below are the details I captured when issuing OFF and ON commands.
1. OFF Command

Request to atCommandv3:

```
{
  "cid": "5162",
  "inverterId": "1308675217947383270",
  "language": "6",
  "localTime": 1753089630408,
  "localTimeZone": 2,
  "value": "222",
  "yuanzhi": "190"
}
```

Corresponding response from atRead:

```
{
  "cid": "5162",
  "inverterId": "1308675217947383270",
  "language": "6",
  "localTime": 1753089566454,
  "localTimeZone": 2,
  "widgetType": 5
}
```

2. ON Command

Request to atCommandv3:

```
{
  "cid": "5162",
  "inverterId": "1308675217947383270",
  "language": "6",
  "localTime": 1753089905833,
  "localTimeZone": 2,
  "value": "190",
  "yuanzhi": "222"
}
```

Corresponding response from atRead:

```
{
  "cid": "5162",
  "inverterId": "1308675217947383270",
  "language": "6",
  "localTime": 1753089891326,
  "localTimeZone": 2,
  "widgetType": 5
}
```

Let me know if you'd like me to capture more examples or test other scenarios. I’m happy to help further if needed.

Thanks again for your time and great work on the integration!

mkuthan, 2025-07-22:
Thanks for providing the HTTP request and response dumps from the Solis webpage. It appears that your inverter (or its firmware) doesn't use 'cids' 52/54 for controlling on/off, but instead uses 'cid' 5162. What's even more interesting is that my own inverter uses 'cid' 6894 on the Solis webpage, while 'cids' 52/54 work perfectly with the Solis Cloud API.

In general official specification defines "cid" 52/54 for controlling on/off in hybrid inverters, look at the "hybrid inverter" sheet in this document:
https://oss.soliscloud.com/doc/SolisCloud_control_api_command_list.xlsx

Could you contact Solis support and ask why your inverter's firmware does not comply with the official Solis Cloud Control API specification?

In the meantime, I will prepare an alpha/beta release modified to use 'cid' 5162 for controlling your inverter.

ElGringo33, 2025-07-24:
Thanks for the clarification — I’ll reach out to Solis support to ask why my inverter uses cid 5162 instead of the standard 52/54 as documented.

Really appreciate that you’re preparing an alpha/beta version using 5162 — happy to test it as soon as it’s available.